OpenStickFoundation/GP2040-CE #15: Additional SOCD options (such as for rhythm games)

Fortinbra, 2022-05-21:


henrebotha, 2022-08-17:
Can we enumerate all the options we want? I know some work is being done here.

nullczek, 2022-08-30:
See my comment in https://github.com/OpenStickFoundation/GP2040-CE/issues/68
This feature could utilize the macro def functionality i discuss therein for the button chords, etc.
Also the game mode profile should be considered https://github.com/OpenStickFoundation/GP2040-CE/issues/77

- per button pair SOCD config - refactor code to allow a per-button pair config of SOCD behavior rather than a tuple for all directionals. 
    - This functionality could really apply to any button pair, not just directionals, but that could also be a macro definition with some options for priority.
    - Can still offer the user pre-canned tuples for users in config for ease of use, but allow per-axis config, and storage
    - allow toggling through cleaning modes for a specific axis through a button combo. This way user toggles through cleaners for any axis up to 5 times to find the mode they want, rather than possibly having to toggle through 25 permutations 
    - display SOCD mode for each axis on oled, possibly flash LEDS in pattern assigned to each mode (ie Neutrai or Max Cardinal)
- Add option to disable SOCD alltogether
- Ability to customize SOCD behavior. This is where the macro definition functionality referenced in my comment on #68 comes into play. A socd cleaner for two buttons is a truth table with 4 rows. Each of these 4 rows can be described using the same methodology i suggested for macros, perhaps with some restrictions on what the user can specify when defining a macro vs a socd mode.
    - for example if @henrebotha wants there a single frame of neutral to be inserted when a clean for last input occurs, he could drag a node representing Neutral input for X milliseconds into each relevant row that corresponds to the truth table.

chikage8640, 2022-11-27:
I want this too.
What I would most like is an option to simply disable SOCD.
Like the Logitech F310, all buttons on the D-pad need to be pressed to get input in all directions. Because in rhythm games there are commands to press all up, down, left and right at the same time.

closed 2022-12-21 by TheTrainGoes