OpenStickFoundation/GP2040-CE #18: Document Developer Setup Process

jbarket, 2022-05-22:
Right now, developing for GP2040 is a nebulous process. We need documentation in the Wiki (with screenshots, preferably) to explain the installation process of platformio, wizio-pico, et cetera.

Bonus points for:

1. WSL2 specific issues, eg passing USB through to prevent having to copy the file
2. Pico Probe documentation (probably a separate ticket, but can't really exist without this one)

NoswalDev, 2022-05-23:
I volunteer to get an outline started this weekend. If someone gets to it sooner, I have no complaints.
My suggestion is that we use Github Pages for a wiki instead of the built-in wiki. That way, we can have version control and allow multiple people to work on it at once without needing elevated access to this main repo.

jbarket, 2022-05-24:
It's funny you say that. I realize there is documentation for this already in the docs here: https://gp2040.info/#/development

Which raises an interesting question. First, I definitely think we want to split end user documentation from dev documentation. But I think the second is, we should make sure to link wherever it goes from the README or wiki here just in case. Because there's so much info there, I didn't even realize it was on the site.

arntsonl, 2022-07-20:
Just a note on this one, wizio broke with the latest version.

So for building the current version, you can use my modified wizio-pico repo as your Wizio in PlatformIO: https://github.com/arntsonl/wizio-pico until the owner gets a fix in.

TheTrainGoes, 2022-12-21:
Closing here, will re-open on the new GitHub once we make the switch to CMake

closed 2022-12-21 by TheTrainGoes