OpenStickFoundation/GP2040-CE #30: Display freezes after a few seconds

butterrsoft, 2022-08-01:
Hi, i have an SSD1306 128x64 OLED display, and it freezes about 10 seconds the stick is plugged in. Hot-plugging the OLED makes it work for another ten seconds or so. 

Also, after powering down the stick, if i plug it back in right away the display rarely comes up. Hot-plugging the OLED does nothing there.

I have a couple more OLEDs on order from another seller and will try to figure out if the display itself is the issue.

butterrsoft, 2022-08-04:
Actually, on longer testing, sometimes the display comes to life again, and freezes again, and comes to life again, and repeat.

TheTrainGoes, 2022-08-04:
Hi @butterrsoft ,

Can you link the specific OLED that you are using?  I have run into some issues with OLEDs from Amazon and AliExpress.

butterrsoft, 2022-08-04:
This one from Aliexpress - https://www.aliexpress.com/item/32233334632.html?spm=a2g0o.order_list.0.0.6a181802U0pJZs

I have a couple more on order that actually say they have the ssd1306 chip in them, but they're probably the same.

TheTrainGoes, 2022-08-05:
Hi @butterrsoft ,

Feel free to join the OpenStick Discord if you would like to have a more in-depth chat about this.  

Can you confirm which IIC address is selected on the back of the OLED and also what your display settings are currently set at in the online config tool?

TheTrainGoes, 2022-08-31:
Closing this as we have not been able to reproduce the issue via any configuration and consider this a hardware issue.  Will re-open if we can reproduce.

TheTrainGoes, 2022-09-01:
We have confirmed that this was a bad OLED module.

closed 2022-09-01 by TheTrainGoes