OpenStickFoundation/GP2040-CE #34: Add extra button to modify inputs like The Antagonist board?

gamecenterjun, 2022-08-10:
Ok so this is not an issue, it's more like a feature request... I'm sure you guys have heard of The Antagonist by now. It's a mod board that adds a button to modify inputs and adds hitbox-like features to lever controllers.

For example, if you hold back and press the extra button, the input changes to forward. Hold back, press button+punch, you get a Sonic Boom without letting go of back, just like a Hitbox. 

It sounds like a feature that should be very useful on the GP2040, so I'm wondering if it would be feasible to add it?

TheTrainGoesChoo, 2022-08-11:
Hey @gamecenterjun ,

I know Buttero very well!  

At this time we are not looking to implement things into the build that are produced by independent makers.  

The Antagonist offers a number of features that we currently do not such as the mobile app for configuration.  

I would recommend grabbing one of his boards, they are great!

TheTrainGoes, 2022-08-11:
Hey @gamecenterjun ,

I know Buttero very well!

At this time we are not looking to implement things into the build that are produced by independent makers.

The Antagonist offers a number of features that we currently do not such as the mobile app for configuration.

I would recommend grabbing one of his boards, they are great!

closed 2022-08-11 by TheTrainGoes