OpenStickFoundation/GP2040-CE #37: RFC: Move board configs out to their own repos

henrebotha, 2022-08-11:
At present, board configs (for specific controllers such as Crush Counter, as well as dev boards such as the Pi Pico) live inside this repo. This means for example that any change to the core code must be tested against every single board, or risk breaking them. It also results in an inverted balance of power, where the owner of a board cannot make changes to their own board without the permission of GP2040-CE maintainers, but the GP2040-CE maintainers can, at any moment, render a board unusable.

So instead, I propose we rearchitect this so that board configs are out of scope for this repo. Doing so allows GP2040-CE to continue developing at a breakneck pace, while leaving it up to individual board owners to decide if and when to upgrade to the latest version. Board owners can elect to pin their board to a particular version of GP2040-CE that works for them.

I'm very inexperienced in the embedded space, so I'm not sure what the technical implications would be.

henrebotha, 2022-08-12:
Here is a good example of how this can be done: https://zmk.dev/docs/user-setup

That project uses GitHub Actions to build the firmware, obviating the need for board designers to install the toolchain. GitHub also has a feature for creating repos based on a template, which would be a great way to automate a lot of the setup for a board designer.

TheTrainGoes, 2022-09-15:
Hey @henrebotha ,

We talked about this a fair bit and landed on this as our stance for the issue:

The main team on GP2040-CE will support only the ongoing development of the main Pico releases.  We will not test or factor other devices into the ongoing development of GP2040-CE.  It is up to the creators of those products to compile the newest releases in a way that works for their devices if they choose.  If they choose to do so we are happy to add it back in as a tested release.

We feel this is a good middle ground that allows us to continue focusing on GP2040-CE as a whole but still allows community products to be included and supported.

closed 2022-09-15 by TheTrainGoes