OpenStickFoundation/GP2040-CE #41: Bug in LS/RS slider when pins are defined as -1 during compile

TheTrainGoes, 2022-08-13:
There is currently a bug where if the LS / RS slider is defined as -1 in the config file the unit will default to LS.  

Stick selection is not persistent though reset so it will always go back to LS even if placed in a different mode via the online config.

There are two ways that I believe we can address this:

1 - Make DPad mode the default, not LS
2 - Force defining of a pin for each the LS / RS slider pins

Additionally, Ferals old documentation talks about a shortcut for selecting the functions via `start` + `select` + `left` for LS, `right` for RS and `up` for DPad, I would like to explore having this added as there is no way to do this currently while in game.

arntsonl, 2022-08-13:
Pins for LS and RS were set to unsigned 8-bit, which cause the comparisons to fail. Added cast on the conversions, will note for future issues.

arntsonl, 2022-08-14:
Fixed in 0.5.1 release.

closed 2022-08-14 by arntsonl