OpenStickFoundation/GP2040-CE #64: flashing buttons showing on OLED screen while holding u + d clean to neutral.

nullczek, 2022-08-17:
hold u + d, with clean to neutral, for about 20 seconds. shortly, on the oled the dpad indicators begin alternating, rather than showing cleaned output of neutral. does not appear to be consistent timing or pattern but not 100% on that. 

See attached movie for demo of behavior. 

this is on 0.5.1, Bentobox using the ScrubTier MPG fork
However, this behavior is NOT observed on a 0.4 build of CE with the same ScrubTier MPG fork. On the 0.4 build correct behavior is observed. 

The UI behavior is a minor issue, but the bigger consideration is whether it's not just a UI issue. If somehow cleaning fails, and the firmware ends up reporting the uncleaned input.  On initial testing in SFV, this does not appear to be the case, but more detailed analysis is warranted.

https://user-images.githubusercontent.com/111413493/185046719-3b2924fb-3e6e-449b-82a6-9d32c4f79610.mov

nullczek, 2022-08-17:
Standard Bento config, with following additions, in case it matters:

#define PIN_BUTTON_TURBO -1
#define PIN_SLIDER_LS    -1
#define PIN_SLIDER_RS    -1

#define BUTTON_LAYOUT BUTTON_LAYOUT_HITBOX
#define BUTTON_LAYOUT_RIGHT BUTTON_LAYOUT_HITBOX

#define SPLASH_MODE NOSPLASH

#define TURBO_LED_PIN -1

#define PLED_TYPE PLED_TYPE_NONE
#define PLED1_PIN -1
#define PLED2_PIN -1
#define PLED3_PIN -1
#define PLED4_PIN -1

TheTrainGoes, 2022-08-17:
Thanks for this.  I was able to reproduce the screen issue.  Going to test on my Switch and see what is registered.

TheTrainGoes, 2022-08-17:
Testing on my Switch does properly clean to neutral, so this is an OLED display issue only.

TheTrainGoes, 2022-08-17:
@nullczek I looked into the code and tested a few things.  This is an issue due to the I2C speed.  

By default the speed is set to 800,000 which is very very fast.  This is causing the issues with the flickering button presses when both are held.  

Lowering this value to 100,000 addresses the issue and shows nothing on screen when both buttons are held as it should.  It has the added bonus of working on the larger 2.42" screens as well.

I will recommend that the default I2C speed be set at 100,000 from now on.

closed 2022-08-18 by arntsonl