OpenStickFoundation/GP2040-CE #68: Add support for macros (perhaps definable in the web interface)

tidle, 2022-08-20:


TheTrainGoes, 2022-08-23:
Macros are high on our want-to-do list!

nullczek, 2022-08-30:
I really think looking around to see if we can borrow code for this from another projects, as this is a very well defined paradigm, 
at the very least see if we reuse a popular data format for persisting macro configs, maybe a webui.

Concepts like layer shifts, button holds/taps multiple presses, chord definition, is well defined in qmk/via, and stored in JSON.

Also, I would suggest checking out reWASD to see how they have in implemented, as it is a pretty powerful system. Uploaded a short video here of reWASD in action.
https://user-images.githubusercontent.com/111413493/187549483-ff49ce09-227e-4ef8-903b-cdbaf7fc3416.mp4

**Importantly**, i think we need to get at least a light implementation of this in place before we tackle things like SOCD customization or definition of hot key button combos for other features like changing game profiles, socd mode, etc.

having a common interface for defining behavior of button chords would be good usability, less code, and all the info can be stored in same format, which will make it easier to avoid having the same chord mapped to two different behaviors.

TheTrainGoes, 2022-12-21:
Macro PoC done.  More to come in the new project

closed 2022-12-21 by TheTrainGoes