OpenStickFoundation/GP2040-CE #69: Weird LED flickering while holding direction inputs

fabricioanciaes, 2022-08-22:
[Here's a video of it happenning](https://imgur.com/a/jbJUS4O)

Pressing the right side buttons and using the static modes where the whole controller lights up work just fine, but for some weird reason, on the mode that the buttons light up when pressed, only the direction buttons flicker. In the video footage you can see me holding all direction buttons and the buttons just flicker like crazy.

This issue does not occur on version `0.4.3`, the version I've experienced this bug is the `0.5.1` release

My `BoardConfig.h` is like this:

```cpp
#ifndef PICO_BOARD_CONFIG_H_
#define PICO_BOARD_CONFIG_H_

#include <GamepadEnums.h>

#define PIN_DPAD_UP     2
#define PIN_DPAD_DOWN   3
#define PIN_DPAD_RIGHT  4
#define PIN_DPAD_LEFT   5
#define PIN_BUTTON_B1   6
#define PIN_BUTTON_B2   7
#define PIN_BUTTON_R2   8
#define PIN_BUTTON_L2   9
#define PIN_BUTTON_B3   10
#define PIN_BUTTON_B4   11
#define PIN_BUTTON_R1   12
#define PIN_BUTTON_L1   13
#define PIN_BUTTON_S1   16
#define PIN_BUTTON_S2   17
#define PIN_BUTTON_L3   18
#define PIN_BUTTON_R3   19
#define PIN_BUTTON_A1   20
#define PIN_BUTTON_A2   21
#define PIN_BUTTON_TURBO -1
#define PIN_SLIDER_LS    -1
#define PIN_SLIDER_RS    -1

#define DEFAULT_SOCD_MODE SOCD_MODE_UP_PRIORITY
#define BUTTON_LAYOUT BUTTON_LAYOUT_STICKLESS
#define BUTTON_LAYOUT_RIGHT BUTTON_LAYOUT_STICKLESSB

#define SPLASH_MODE CLOSEIN

#define TURBO_LED_PIN 15

#define BOARD_LEDS_PIN 22

#define LED_BRIGHTNESS_MAXIMUM 128
#define LED_BRIGHTNESS_STEPS 5
#define LED_FORMAT LED_FORMAT_GRB
#define LEDS_PER_PIXEL 2

#define LEDS_DPAD_LEFT   0
#define LEDS_DPAD_DOWN   1
#define LEDS_DPAD_RIGHT  2
#define LEDS_DPAD_UP     3
#define LEDS_BUTTON_B3   4
#define LEDS_BUTTON_B4   5
#define LEDS_BUTTON_R1   6
#define LEDS_BUTTON_L1   7
#define LEDS_BUTTON_B1   8
#define LEDS_BUTTON_B2   9
#define LEDS_BUTTON_R2   10
#define LEDS_BUTTON_L2   11

#define PLED_TYPE PLED_TYPE_NONE
#define PLED1_PIN -1
#define PLED2_PIN -1
#define PLED3_PIN -1
#define PLED4_PIN -1

#define ANALOG_ADC_VRX -1
#define ANALOG_ADC_VRY -1

#define HAS_I2C_DISPLAY 0
#define I2C_SDA_PIN 0
#define I2C_SCL_PIN 1
#define I2C_BLOCK i2c0
#define I2C_SPEED 400000

#endif
```

nullczek, 2022-08-22:
SOCD flashing on --OLED-- screen still present at 100, 400 k I2C speed, likely related to #64

TheTrainGoes, 2022-08-23:
Thank you for this - we are looking into it.

arntsonl, 2022-08-28:
Solved it, the issue was that the add-ons could read the gamepad while it was in the middle of cleaning SOCD.

I've added a solution that works and is pretty easy, but we will want to rethink MPG a bit.

TheTrainGoes, 2022-08-28:
Hi @fabricioanciaes ,

Feel free to compile https://github.com/OpenStickFoundation/GP2040-CE/pull/75 and test this out for yourself if you have any additional feedback.  

I have tested this and it has fixed the issue.

closed 2022-08-28 by TheTrainGoes