OpenStickFoundation/GP2040-CE #76: Implement tournament lockout mode

nullczek, 2022-08-30:
Accidentally pressing a button such as start during tourney play is grounds for a DQ.
Lockout harnesses use a toggle switch to disable such buttons.

Could allow this to optionally be mapped to a pin, or do in software, where it can be toggled, optimally through a combination hold for X seconds, possibly two such combos to eliminate possibility of accidental change.

Enabling (not just turning it on or off) the feature should be a toggle in webconfig, as people may not want to deal with it in non FGC scenarios.
If feature is enabled, an off/on toggle.  
options for configuring which buttons are locked out 
options for configuring the button combos, but need to think about how to avoid having "magic button" mappings all over the config, running the risk of creating same button combo for other features. More on that in my next Issue.

a clear method of communicating the state of the lockout is needed. OLED persistent message with status. if no OLED, maybe an option for changing color of LEDs on a set of buttons. 
Ex: 
if tournament mode lockout feature disabled
   LED color settings apply as today.
else if tournament mode lockout  feature enabled, 
   if tournament mode lockout setting is off
      make (set of buttons) red
   if tournament mode lockout  setting is on
      make (set of buttons) green

Adjustable colors for the colorblind,

If no oled or leds, then they just have to test and be real sure.

nullczek, 2022-08-30:
See my comment in #68
This feature could utilize the macro def functionality i discuss therein for the button chords, etc.

TheTrainGoes, 2022-08-31:
Hi @nullczek ,

Currently tournament legality is considered out of scope for the GP2040-CE project.  

We are focused on getting to a stable v1.0 before looking into the overall legality of GP2040-CE in tournaments.

The simple solution to any sort of lockout is still to have a physical switch on the stick itself. 

I like the proposed idea of having an OLED based indication of what is and is not locked if done though software.  This however relies on the person having a case that accommodates an OLED.  

I am unsure about using LEDs to communicate the state of the lockout as I have read mixed things about the use of LEDs during tournament play.  

For now we are going to remain focused on the GP2040-CE project without tournament legality in mind.  I am reaching out to a few organizers to talk about GP2040 / RP2040 as a whole to see what is and is not possible.

Fortinbra, 2022-08-31:
I'm closing this issue for now. If we change our stance, we can always reopen or create a new issue.

closed 2022-08-31 by Fortinbra

henrebotha, 2022-09-05:
Despite the name, this feature is not restricted to the tournament scene. It should not be wontfix just because of its name.

TheTrainGoes, 2022-09-05:
@henrebotha I am still more in favour of a physical switch that just cuts the ground line.  

For builds without the screen / LEDs it would need a key combo that is available on the non-locked out buttons to enable / disable.  

This makes more sense when there is a screen that shows things are locked out, but I like to try and work features around the ensure user base and the creations they may already have.

henrebotha, 2022-09-05:
> it would need a key combo that is available on the non-locked out buttons

I don't think this is necessary. In fact it might be quite convenient to use the locked-out buttons themselves, in a more strict way (for example, requiring the player to hold two buttons down for 3 seconds). That way you don't require extra buttons just to control the feature. 

Hard switches are good, but it's nice to have the option to implement this in firmware.