OpenStickFoundation/GP2040-CE #87: LED flickering when connected to certain devices.

SneakySpec, 2022-09-03:
LEDs rapidly cycle through colors and flash when a controller is connected to specific devices.

[Video Example 1](https://www.youtube.com/shorts/RUzpTee3nq8) -  0.4.3

(brightness and flashing warning)
[Video Example 2](https://www.youtube.com/shorts/RUzpTee3nq8) -  0.5.2

This behavior has been found to be exhibited when connected to the following devices.

Thunderbolt 3 Port
Playstation 3
Playstation 4

Toggling X-input/D-input/Switch mode does not alter the behavior. LEDs glitch out on the aforementioned devices, but  behave normally on other platforms.

On 0.5.2 this behavior is consistent across all LED modes and settings. On 0.4.3 it has the same issue, but behaves slightly differently.

(brightness and flashing warning)
[Video Example 3](https://streamable.com/hpv5d8) -  0.4.3

The chasing animation plays out, but it stutters and each LED rapidly cycles through colors. The static themes have some degree of stability and only show brief, intermittent flickering. The static rainbow theme is the most stable with little to no flickering.  The static themes will start to destabilize as the brightness is decreased. Low enough brightness settings will present behavior consistent with what occurs on 0.5.2 in all LED modes.

TheTrainGoes, 2022-09-06:
Thank you for the report @SneakySpec ,

We will take a look at this when we have time and I will see if I can replicate on my own.

TheTrainGoes, 2022-11-13:
Hey @SneakySpec,

I went out and bought a PS3 and am not able to replicate this on mine.  

I have tried a few different RP2040 setups and all are displaying the LEDs as expected.

Can you please let me know what device you are having this issue with, what mode it is, which plug is being used / etc?

TheTrainGoes, 2022-11-27:
Going to close this as unable to reproduce.  @SneakySpec if you can get in touch with me via Discord we can talk more about it.  

I was watching the video again and am wondering if you have some power delivery issues on the 5v line causing this.

closed 2022-11-28 by TheTrainGoes