OpenStickFoundation/GP2040-CE #88: Allow mapping all analog stick inputs simultaneously

TheOpponent, 2022-09-04:
Mapping left and right analog stick directions to the pins would allow for control of games that use them simultaneously, such as twin-stick shooter games.

TheTrainGoes, 2022-09-06:
Hi @TheOpponent ,

We will take a look at the feasibility of this.

TheTrainGoes, 2022-09-22:
@TheOpponent,

Wanted to check in on this again to make sure I am understanding the ask.  

Are you asking for a single analog input to be emulated as the other analog input simultaneously, or are you just asking to have two separate analogue sticks that can be used independently at the same time?

TheOpponent, 2022-09-22:
My request is for the latter, allowing digital inputs to be mapped to both left and right analog controls at once. Currently, the directional input mode can be switched between the D-pad, left analog, or right analog, and I would like to know if it would be viable to allow left and right analog input at the same time, all controlled with digital inputs. My use case is for two 5-pin joysticks to be used in twin-stick games, and all such games made today expect analog input.

TheTrainGoes, 2022-09-22:
@TheOpponent thanks for the update.

Twinstick is something I have looked into as well as I want to make one myself.  

Most games allow for remapping to other inputs for the 2nd stick.  I built a proof-of-concept for the Switch that used the D-Pad and buttons as the separate sticks.  

I will bring this up internally to see if there is a way to emulate RS via P1, P2, K1, K2.

TheTrainGoes, 2022-09-22:
Talked internally.  

Not impossible, but also not currently on the roadmap.  

Will close this out as not planned but keep it on our minds for future consideration.

TheOpponent, 2022-09-22:
> Most games allow for remapping to other inputs for the 2nd stick. I built a proof-of-concept for the Switch that used the D-Pad and buttons as the separate sticks.
Many of the twin-stick games I play are on PC, and they don't allow mapping buttons for the aiming controls. They only allow aiming with either the mouse or an analog stick input.

Thank you for your consideration.

closed 2022-09-22 by TheOpponent