OpenStickFoundation/GP2040-CE #89: Wrong LED order when inverting DPAD mode

endl, 2022-09-05:
when using inverted DPAD if the down button is pressed the LED for the up button is illuminated and visa versa
video of bug:
https://photos.app.goo.gl/UAk4NvH8KEwhtYp29

TheTrainGoes, 2022-09-06:
To elaborate on this my suspicion is that when the invert DPAD shortcut is used it is not also switching the assigned RGB LEDs order.

```[Invert D-Pad Y-axis](https://gp2040.info/#/usage?id=invert-d-pad-y-axis)
A toggle is available to invert the Y-axis input of the D-pad, allowing some additional input flexibility. To toggle, press START + HOME + RIGHT. This is a temporary hotkey mapping for this feature, so keep an eye on updated releases for this to change.```

TheTrainGoes, 2022-09-06:
Further looking into this we will need to make changes to MPG which so far we have done done.  

This may be out-of-scope for the time being but we will leave it open.

closed 2022-12-21 by TheTrainGoes