OpenStickFoundation/GP2040-CE #90: Error when building from source

Freamerhxl, 2022-09-06:
Uploading...
  Converting to UF2 ( 0x10000000 )
  Wrote 2161664 bytes to E:/source/GP2040-CE/.pio/build/raspberry-pi-pico\APPLICATION.uf2
*** [upload] UnicodeDecodeError : 'utf-8' codec can't decode byte 0xd0 in position 129: invalid continuation byte
Traceback (most recent call last):
  File "C:\Users\Fream\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\Fream\.platformio\platforms\wizio-pico\builder\frameworks\uf2conv.py", line 337, in dev_uploader
    drives = get_drives()
  File "C:\Users\Fream\.platformio\platforms\wizio-pico\builder\frameworks\uf2conv.py", line 201, in get_drives
    for line in to_str(r).split('\n'):
  File "C:\Users\Fream\.platformio\platforms\wizio-pico\builder\frameworks\uf2conv.py", line 195, in to_str
    return b.decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 129: invalid continuation byte

My env
windows11
python3.9

arntsonl, 2022-09-13:
This looks like a Platform.Io issue (scons-local-4.4.0 specifically), as we do not do anything in the code at this point.

What Windows 11 language / localization are you set to?

Wang-Jianguo, 2022-09-13:
1. ‘Use Unicode UTF-8 for worldwide language support’ is enabled in Windows Region settings

arntsonl, 2022-09-13:
Perfect thank you for the info.

Yeah this is part of wizio-pico, which we can't currently update as it breaks LWIP. I'll put some more time into seeing if there's anything we can do about this.

TheTrainGoes, 2022-12-21:
Closing here, will re-open on the new GitHub if still wanted

closed 2022-12-21 by TheTrainGoes