OpenStickFoundation/GP2040-CE #94: Nintendo Switch mode is not recognized on REAL Switch

xboxgod, 2022-09-18:
set to switch mode and windows system display POKKEN CONTROLLER, but insert to nintendo switch machine is not recognized, like no controller is plugged in. this is a BUG ?

TheTrainGoes, 2022-09-19:
Hey @xboxgod,

I am primarily a Switch user myself and can confirm that it works 100%.

If not already done, you need to enable pro controller wired communication.  You can find steps to that here:

https://en-americas-support.nintendo.com/app/answers/detail/a_id/26315/~/how-to-enable%2Fdisable-pro-controller-wired-communication#:~:text=Complete%20these%20steps,it%20again%20to%20disable%20it.

"Complete these steps:
1 - From the HOME Menu select System Settings, then select Controllers and Sensors.
2 - Select Pro Controller Wired Communication to enable this feature. Select it again to disable it."

You also will need to hit a button on the Pico based controller for it to show up.

xboxgod, 2022-09-19:
I can recognize it when I plug it in Switch with the same OTG adapter 
The Brook Fighting Board, is OK  but 0.5.2 has no response at all

I will try, thanks a lot

TheTrainGoes, 2022-09-20:
Worked with the user directly.  Wired Pro Controller setting was not turned on.  Issue resoled.

closed 2022-09-20 by TheTrainGoes

TheTrainGoes, 2022-09-20:
As an additional note - modded Switches and Switches that are running old firmware from before the introduction of Wired Pro Controller mode may not work as expected.