OpenStickFoundation/GP2040-CE #98: I also have this problem. How can I solve it?

BDJoe360, 2022-09-21:
1.AS:
https://github.com/OpenStickFoundation/GP2040-CE/issues/94#issuecomment-1252650080
My switch is a version 14.2 system that has been modified by the atmospheric soft magic.Is there a solution?

2.In addition, my Pico is installed on Pico Fighting Board v1.1. The original firmware can only use GP2040 PicoFightingBoard_ V0.4.3. uf2, GP2040 RaspberryPiPico_ The key position of v0.4.3. uf2 is abnormal.The same situation occurs in the two firmware.I wonder if you can check and correct it?

TheTrainGoes, 2022-09-22:
HI @BDJoe360 ,

Are you able to turn on Pro Wired Controller mode from the settings still?

Hacked or modified Switches are touch to troubleshoot.

What is the behaviour when you connect it in switch mode and then press the A button multiple times?

BDJoe360, 2022-09-22:
> HI @BDJoe360 ,
> 
> Are you able to turn on Pro Wired Controller mode from the settings still?
> **Yes,I can enable Pro Wired Controller mode from the settings.In addition, the original Pro Controller and 8-bitdo GBros can be used normally. (whether wired or wireless)**

> Hacked or modified Switches are touch to troubleshoot.
> **I don't quite understand. What do you mean?**

> What is the behaviour when you connect it in switch mode and then press the A button multiple times?
>**No reaction, completely unrecognizable.But when plugged into a PC, it will be recognized as POKKEN CONTROLLER (Vendor: 0f0d Product: 0092)**

**in addition,Can you increase support for Xbox 360 game console?**

TheTrainGoes, 2022-09-22:
Hi @BDJoe360 ,

Support for modded or hacked Nintendo Switches is going to be considered out of scope.  There are no issues with non-modified or hacked Switches.  The modding or hacking process changes internal files which may result in GP2040-CE not working as intended.   

  At this time there is no support for xBox / PS4 / PS5.  If we are able to ever get a legitimate way of offering support for those devices we will scope out its implementation.

closed 2022-09-22 by TheTrainGoes