OpenStickFoundation/GP2040-CE #102: i2cdisplay fails to render several directional styles resulting in blank screen

nullczek, 2022-09-29:
for #define BUTTON_LAYOUT
if the following values are used, the end result is a blank screen, and no LEDs on the buttons

BUTTON_LAYOUT_BUTTONS_BASIC
BUTTON_LAYOUT_KEYBOARD_ANGLED
BUTTON_LAYOUT_KEYBOARDA
BUTTON_LAYOUT_DANCEPADA

TheTrainGoes, 2022-09-30:
Strange, I had tested these at one point.  Will check them out tomorrow.

TheTrainGoes, 2022-10-01:
Testing this today

TheTrainGoes, 2022-10-01:
@nullczek I have tested all of these and they are working on my setup.  

Can you tell me a bit more about what you have setup wise to replicate this?

TheTrainGoes, 2022-10-08:
Going to close this as resolved as I have not been able to replicate.  If we can replicate we can take a look again.

closed 2022-10-08 by TheTrainGoes

nullczek, 2022-10-11:
This was using the default Bento board config.
When using all of the the left side defines referenced as defective in this
issue, I’ve tried the following:
1. Disabling the rendering of the right hand side buttons altogether
2. Between the referenced left hand defines, I’ve tried using all possible
right side button defines

The outcome was always the same - when referenced left side values are
used, the oled is blank and the LEDs on the buttons do not come on.

Not sure if you ever managed to get a Bento, might be good to have
ScrubTier confirm behavior.