OpenStickFoundation/GP2040-CE #104: Support for multiple buttons to same input

Hoopyy, 2022-09-29:
For instance, I would like to have 'Down' bound to down on my lever, the A button on my stick and the L2 button on my stick (3 ways total to press down). Trying to do this through the web-config results in an error. This wouldn't be a permanent configuration for my stick though, as I'd like to change input layout for different games (ruling out hardwiring buttons to same inputs).

I guess by extension it'd be cool to have a save/load config option in the web-config for easy switching between layouts.

TheTrainGoes, 2022-09-30:
Hey @Hoopyy ,

Thanks for the feedback.  

Can I compile something and send it over for testing to see if it behaves as expected and then we can look into the limitations for the config file.

Save / load is currently out of scope.

Hoopyy, 2022-09-30:
Yeah, I'd be happy to test.

TheTrainGoes, 2022-10-01:
I've been testing out a few things on this but when defining in the BoardConfig.h you can only specify each input once.  So even if you put multiple down it will only take the last one.

Since this is a limitation to the code it is not something changing web-config would fix.  

For this I think your best bet would be some custom hardware that can allow for multiple buttons to be mapped to the same pin.  I have this done on my Retro Pocket Pad via a switch that you can do in real time.

If you want to give me a more broad breakdown of all the changes you might want to make I can look into makeing you a custom board for it.

For now though we will consider this out-of-scope with no plans to do until we decicde to really start cutting up MPG and breaking some of the limitations on its current implementation.

closed 2022-10-01 by TheTrainGoes

Hoopyy, 2022-10-01:
Alright no problem. Biggest reason for wanting this functionality was due to the Steam controller configuration adding unnecessary input delay (I believe its 1ms, very minor, but its extra delay none-the-less). I've found other programs that serve the same purpose with no delay, so I can use those. No need to bother yourself creating a custom board just for my use case. 

Thanks for looking into this issue

nullczek, 2022-10-01:
At a minimum I’d like to take a look at MPG, think about what it would take
for a minimum viable product implementation, and add the notes here. Please
assign to me so I don’t lose track of this.

TheTrainGoes, 2022-10-02:
Assigned to you @nullczek