OpenStickFoundation/GP2040-CE #109: Detach GP2040-CE as fork of FeralAI/GP2040

davewongillies, 2022-10-02:
Now that it seems that [FeralAI/GP2040](https://github.com/FeralAI/GP2040/issues/87) is no longer maintained and is archived, can I suggest that you [detach](https://support.github.com/request/fork) this repository from being a fork of it? I've noticed that since this is a fork, it seems that it gets put down pretty low in search rankings on both Google and Github. In the past I maintained an open source project which started life as an abandoned fork, and after detaching it from the fork, I found that it was easier for people to find it.

TheTrainGoes, 2022-10-02:
Tagging @Fortinbra on this one as they look after the this part of things

davewongillies, 2022-12-21:
I guess this can be closed now that things have moved over to https://github.com/OpenStickCommunity/GP2040-CE

closed 2022-12-21 by davewongillies