OpenStickFoundation/GP2040-CE #138: 0.5.2 - Impossible to access webUI

generalerusso, 2022-11-04:
Recently installed the GP2040-CE-RaspberryPiPico_v.0.5.2_Stick.uf2 on a brand new Pi Pico (non W) and while it works perfectly fine, I can't seem to access the webUI at 192.168.7.1. I have followed the pinout, so i do have the START button connected to the pin S2 (GP17).

When I connect the USB cable to my PC while pressing the START/S2 button, it does appear to change its ID when using `lsusb`, and it appears with `ID cafe:4001`, instead of the normal mode with the generic HID controller `ID 0f0d:0094`.

I'm on Linux Mint 19.3, I've tried connecting to the webUI with Firefox browser and Epiphany browser, but to no avail, the connection will eventually timeout. I've also tried to connect to that IP while disconnected from the internet, but the browser won't even start the search as "network is not present"

Any comment will be greatly appreciated.

davewongillies, 2022-11-04:
@generalerusso unfortunately access to the webUI is currently only supported on windows. There's an existing issue for it https://github.com/OpenStickFoundation/GP2040-CE/issues/10

TheTrainGoes, 2022-11-14:
Closing this out!  We hope to have additional support for more operating systems soon

closed 2022-11-14 by TheTrainGoes