OpenStickFoundation/GP2040-CE #146: GP2040-CE  as HID for SNES, NES ,  PC  and other controllers...

goyetus, 2022-11-16:
Hello!!   Im about two weeks searching info  to use the RP2040  as a stick HID to control SNES pads....
Im not a low level coder,  so I can not convert NES code to SNES using PICO.

I think it will be awesome to have different controllers implemented inside GP2040-CE.  
Anyone skilled to do it?  

I have samples from :  

NES to USB   using PICO
https://github.com/printnplay/Pico-MicroPython/blob/main/NES2USB.py

PC-DSUB to USB  using PICO
https://www.instructables.com/Retro-9-Pin-Joystick-to-USB-Converter/

SNES to USB  using ARDUINO.....
https://dragaosemchama.com/2015/11/controller-de-super-nintendo-no-pc/

Here all my data search   (have a lot more code with SNES to USB using arduino)

THANKS A LOT FOR YOUR HARD WORK !!!!
GP2040 IS AWESOME !!!!!

goyetus, 2022-11-17:
Before closing this Thread,   PLEASE CONSIDER THIS: 

DaemonBite for arduino SUPPORTS : 
- Arcade Sticks 
- C64 controllers
- Amiga controllers
- Atari controllers
- NeoGeo  controllers

This request is a hard request.  Please Consider this well in the  "planning route"  for GP2040-CE.

[DaemonBite](https://github.com/MickGyver/DaemonBite-Arcade-Encoder)

davewongillies, 2022-11-18:
> Before closing this Thread, PLEASE CONSIDER THIS:
>
> DaemonBite for arduino SUPPORTS :
>
> * Arcade Sticks
> * C64 controllers
> * Amiga controllers
> * Atari controllers
> * NeoGeo  controllers
>
> This request is [...]

GP2040-CE already has a drop in replacement configuration for the Daemonbite Arcade Encoder with [Sparkfun Pro Micro - RP2040](https://github.com/OpenStickFoundation/GP2040-CE/tree/main/configs/SparkFunProMicro) support. The pin mappings are 1-to-1 for arcade sticks.

TheTrainGoes, 2022-11-27:
Closing this as out of scope

closed 2022-11-27 by TheTrainGoes

epid, 2022-12-12:
The NES and SNES controllers don't use a 1-to-1 pin mapping for inputs. They use a data/latch/clock system using 3 I/O pins to represent 12 possible inputs. See here: [https://gamesx.com/controldata/snesdat.htm](https://gamesx.com/controldata/snesdat.htm)