OpenStickFoundation/GP2040-CE #147: Add ability to add custom LED order matrix

waterbury, 2022-11-17:
I have an LED matrix PCB with 4 LEDs per button. The LEDs aren't grouped in button order. I had some code on another microcontroller to deal with this. Please refer to the code block below. Also, using this scheme perhaps allow a low current version that only uses 1 LED from each button.

```
//Holds Button 1-20 LED location. Order NW, NE, SW, SE. ie. buttonLEDs[0] houses position of button 1 NW LED, buttonLEDs[4] houses position of button 2 NW LED. LED starts at position 1.
int buttonLEDpos[80] = { 25, 26, 24, 23    ,    9, 10, 8, 7   ,   27, 28, 22, 21    ,   11, 12, 6, 5    ,
                       29, 30, 20, 19    ,   13, 14, 4, 3   ,   31, 32, 18, 17    ,   15, 16, 2, 1 };
```

waterbury, 2022-11-20:
This change to neopicoleds.cpp works as a hack. I'll try to implement an better change tomorrow and submit
Line 334: 
`
std::vector<std::vector<Pixel>> NeoPicoLEDAddon::createLEDLayout(ButtonLayout layout, uint8_t ledsPerPixel, uint8_t ledButtonCount)
{
	vector<vector<uint8_t>> positions(ledButtonCount);
	int buttonLEDpos[80] = { 25, 26, 24, 23    ,    9, 10, 8, 7   ,   27, 28, 22, 21    ,   11, 12, 6, 5    ,
                       29, 30, 20, 19    ,   13, 14, 4, 3   ,   31, 32, 18, 17    ,   15, 16, 2, 1 };
	/*
	for (int i = 0; i != ledButtonCount; i++)
	{
		positions[i].resize(ledsPerPixel);
		for (int l = 0; l != ledsPerPixel; l++)
			positions[i][l] = (i * ledsPerPixel) + l;
	}*/

	for (int i = 0; i != ledButtonCount; i++)
	{
		positions[i].resize(ledsPerPixel);
		for (int l = 0; l != ledsPerPixel; l++)
			positions[i][l] = buttonLEDpos[ (i * 4) + l ] - 1;
	}	
`

TheTrainGoes, 2022-12-21:
Hey @waterbury ,

Would you mind re-submitting this on the new GitHub?
https://github.com/OpenStickCommunity/GP2040-CE/issues/

closed 2022-12-21 by TheTrainGoes