OpenStickFoundation/GP2040-CE #179: This GitHub is now archived - please check out https://github.com/OpenStickCommunity/GP2040-CE

TheTrainGoes, 2022-12-21:


justin-pierce, 2023-01-09:
link for convenience: https://github.com/OpenStickCommunity/GP2040-CE

darrenoc3, 2023-02-03:
Again? Why? 

Anyway, there's a Github feature for archiving repos that will show a warning to visitors in the page header (rather than in the README as it is currently). Probably worth activating it 
[image]

TheTrainGoes, 2023-02-03:
Hey @darrenoc3 ,

The issue at the time we none of the people currently active on the project had access to make this change.  

When we moved over to the new repo we ensured that there were multiple people with the necessary level of ownership to avoid this happening again (to the best of our ability).  

Having said that one of those people is now back in off and on contact so I will see if we can get it done.